SEAME-pt/DistributedEmbeddedSystems #4: CAN

matde-je, 2024-11-18:


matde-je, 2024-11-26:
We need to use the 2 CANs one for raspberry other for the jetson, so that they can communicate through the CAN protocol. So that the raspberry can communicate the speed to the jetson nano.

matde-je, 2024-11-26:
Its possible we dont need CAN if we are not using raspberry

matde-je, 2024-12-03:
needs to reboot

matde-je, 2024-12-04:
https://www.waveshare.com/wiki/RS485_CAN_for_Jetson_Nano

JoelenCruz, 2024-12-05:
| **Etapa**                           | **Comando**                                       | **Descrição**                                                                 |
|-------------------------------------|---------------------------------------------------|-------------------------------------------------------------------------------|
| **1. Verificar dispositivos SPI**   | `ls /dev/spi*`                                    | Lista os dispositivos SPI disponíveis.                                        |
| **2. Verificar inicialização do CAN** | `dmesg | grep mcp251x`                           | Mostra logs do sistema relacionados ao MCP2515.                              |
| **3. Ativar interface CAN**         | `sudo ip link set can0 up type can bitrate 500000`| Ativa a interface CAN em 500 kbps.                                           |
| **4. Verificar status da interface**| `ip link show can0`                               | Mostra o estado da interface CAN.                                            |
| **5. Configurar modo loopback**     | `sudo ip link set can0 down`                     | Desativa a interface CAN.                                                    |
|                                     | `sudo ip link set can0 up type can bitrate 500000 loopback on`| Ativa a interface CAN no modo loopback.                           |
| **6. Monitorar o barramento CAN**   | `candump can0`                                    | Escuta mensagens no barramento CAN.                                          |
| **7. Enviar mensagem CAN**          | `cansend can0 123#DEADBEEF`                      | Envia uma mensagem CAN com ID 123 e dados `DEADBEEF`.                        |
| **8. Recarregar drivers do MCP2515**| `sudo ip link set can0 down`                     | Desativa a interface CAN.                                                    |
|                                     | `sudo rmmod mcp251x`                              | Remove o driver do MCP2515.                                                  |
|                                     | `sudo rmmod spi_bcm2835`                         | Remove o driver SPI.                                                         |
|                                     | `sudo modprobe spi_bcm2835`                      | Recarrega o driver SPI.                                                      |
|                                     | `sudo modprobe mcp251x`                          | Recarrega o driver do MCP2515.                                               |
|                                     | `sudo ip link set can0 up type can bitrate 500000`| Ativa novamente a interface CAN.                                             |
| **9. Verificar resistores de terminação** | -                                             | Certifique-se de que há resistores de 120 ohms em cada extremidade do barramento. |
| **10. Verificar conexões físicas**  | -                                                | Confirme as conexões entre CANH, CANL, VCC e GND.                            |