SEAME-pt/DistributedEmbeddedSystems #11: Arduino vs jetson nano

matde-je, 2024-11-20:


JoelenCruz, 2024-11-20:
[image]

JoelenCruz, 2024-11-20:
[LM393 Velocity Sensor for Jetson Nano in forum] (https://forums.developer.nvidia.com/t/lm393-velocity-sensor-for-jetson-nano/84279) https://forums.developer.nvidia.com/t/lm393-velocity-sensor-for-jetson-nano/84279

JoelenCruz, 2024-11-20:
# **SEN-Speed LM393 Sensor com Jetson Nano**

Este guia detalha como conectar e configurar o sensor **SEN-Speed LM393** com o **Jetson Nano** para medir a velocidade de rotação utilizando o disco encoder.

---

## **Descrição do Projeto**

O sensor **LM393** detecta a rotação do disco encoder e envia pulsos digitais correspondentes ao número de interrupções ópticas. Esses pulsos podem ser capturados pelo Jetson Nano para calcular a velocidade em **pulsos por segundo** ou **RPM (rotações por minuto)**.

---

## **Materiais Necessários**

- Sensor SEN-Speed LM393
- Disco encoder
- Jetson Nano (com headers GPIO J41)
- Cabos jumper para conexões
- Fonte de alimentação para o Jetson Nano

---

## **Esquema de Conexão**

Conecte os pinos do sensor LM393 ao cabeçalho GPIO **J41** do Jetson Nano conforme abaixo:

| **Pino do Sensor** | **Pino no Jetson Nano (J41)** | **Descrição**             |
|---------------------|-------------------------------|---------------------------|
| **VCC**            | 5V (pino 2 ou 4)             | Alimentação do sensor     |
| **GND**            | GND (pino 6, 9, etc.)        | Terra                     |
| **DO**             | GPIO (ex.: pino 18 / GPIO 12)| Saída digital do sensor   |

> **Nota:** Certifique-se de que o sensor esteja alimentado corretamente (geralmente 5V). Verifique o número de ranhuras no disco encoder para ajustar os cálculos de RPM, se necessário.

---

## **Instalação de Software**

### **1. Instale a biblioteca GPIO**

No Jetson Nano, use a biblioteca **Jetson.GPIO** para acessar os GPIOs:

```bash
pip3 install Jetson.GPIO

JoelenCruz, 2024-11-20:
https://jetsonhacks.com/nvidia-jetson-nano-j41-header-pinout/ pin map of the jetson nano.. on the board there are also markings

closed 2024-11-28 by matde-je