SEAME-pt/DistributedEmbeddedSystems #45: OpenCV

matde-je, 2025-01-27:


matde-je, 2025-01-27:
import cv2
import numpy as np

# Função para detectar lanes
def detect_lanes(image):
    # Redimensionar a imagem para acelerar o processamento
    resized_image = cv2.resize(image, (640, 360))  # Ajuste conforme necessário

    # Converter para escala de cinza
    gray = cv2.cvtColor(resized_image, cv2.COLOR_BGR2GRAY)

    # Aplicar um Gaussian Blur para reduzir o ruído
    # Usando filtro com CUDA
    gpu_gray = cv2.cuda_GpuMat()
    gpu_gray.upload(gray)
    blurred_gpu = cv2.cuda_GaussianBlur(gpu_gray, (5, 5), 0)
    blurred = blurred_gpu.download()

    # Detecção de bordas com Canny (acelerado por CUDA)
    gpu_blurred = cv2.cuda_GpuMat()
    gpu_blurred.upload(blurred)
    edges_gpu = cv2.cuda_Canny(gpu_blurred, 50, 150)  # Ajuste os thresholds conforme necessário
    edges = edges_gpu.download()

    # Máscara para a região de interesse (ROI)
    height, width = edges.shape
    mask = np.zeros_like(edges)
    polygon = np.array([[
        (0, height), (width, height), (width // 2, int(height * 0.6))
    ]], np.int32)
    cv2.fillPoly(mask, polygon, 255)
    masked_edges = cv2.bitwise_and(edges, mask)

    # Detecção de linhas com Hough Transform (tradicional, CPU)
    lines = cv2.HoughLinesP(masked_edges, 1, np.pi / 180, threshold=50, minLineLength=40, maxLineGap=100)
    line_image = np.zeros_like(resized_image)

    # Desenhar as linhas detectadas
    if lines is not None:
        for line in lines:
            x1, y1, x2, y2 = line[0]
            cv2.line(line_image, (x1, y1), (x2, y2), (0, 255, 0), 5)

    # Combinar as linhas detectadas com a imagem original
    final_image = cv2.addWeighted(resized_image, 0.8, line_image, 1, 1)
    return final_image

# Carregar o vídeo (ou webcam)
cap = cv2.VideoCapture("video_lanes.mp4")  # Substitua pelo caminho do seu vídeo ou use `0` para webcam

while cap.isOpened():
    ret, frame = cap.read()
    if not ret:
        break

    # Detectar lanes
    lanes = detect_lanes(frame)

    # Exibir o resultado
    cv2.imshow("Lane Detection", lanes)

    # Sair com a tecla 'q'
    if cv2.waitKey(1) & 0xFF == ord('q'):
        break

cap.release()
cv2.destroyAllWindows()

matde-je, 2025-01-27:
```python
import cv2
import numpy as np
import tensorflow as tf

# Step 1: Capture a frame
cap = cv2.VideoCapture(0)
ret, frame = cap.read()

# Step 2: Preprocess with OpenCV
gray = cv2.cvtColor(frame, cv2.COLOR_BGR2GRAY)
edges = cv2.Canny(gray, 50, 150)

# Step 3: Resize for TensorFlow model
input_frame = cv2.resize(edges, (160, 80))
input_frame = input_frame / 255.0  # Normalize
input_frame = input_frame[np.newaxis, :, :, np.newaxis]

# Step 4: Predict with TensorFlow model
model = tf.keras.models.load_model('lane_detection_model')
predicted_mask = model.predict(input_frame)

# Step 5: Postprocess with OpenCV
output = (predicted_mask[0] > 0.5).astype(np.uint8) * 255
output_resized = cv2.resize(output, (frame.shape[1], frame.shape[0]))
overlay = cv2.addWeighted(frame, 0.8, cv2.cvtColor(output_resized, cv2.COLOR_GRAY2BGR), 0.2, 0)

cv2.imshow("Lane Detection", overlay)
cv2.waitKey(0)
cap.release()
cv2.destroyAllWindows()
```

JoelenCruz, 2025-01-27:
https://forums.developer.nvidia.com/t/okdo-jetson-nano/300923 .. problemas ao tentar conectar a camera usando opencv

JoelenCruz, 2025-01-28:
sudo apt update
sudo apt install -y build-essential cmake git libgtk2.0-dev pkg-config \
    libavcodec-dev libavformat-dev libswscale-dev python3-dev python3-numpy
git clone https://github.com/opencv/opencv.git
git clone https://github.com/opencv/opencv_contrib.git
cd opencv
mkdir build
cd build
cmake -D CMAKE_BUILD_TYPE=Release \
      -D CMAKE_INSTALL_PREFIX=/usr/local \
      -D OPENCV_EXTRA_MODULES_PATH=../../opencv_contrib/modules \
      -D WITH_CUDA=ON \
      -D ENABLE_FAST_MATH=ON ..
make -j$(nproc)
sudo make install

JoelenCruz, 2025-01-30:
- Install python 3.10
- python 3.10 default python
- Install open cv

matde-je, 2025-01-31:
```python
import cv2
import numpy as np
import torch
from cv2.cuda import createGaussianFilter, createCannyEdgeDetector, resize

# Load PyTorch model
device = torch.device("cuda" if torch.cuda.is_available() else "cpu")
model = torch.jit.load('lane_detection_model.pt')  # Use TorchScript model for inference
model.to(device)
model.eval()

# Open the camera
cap = cv2.VideoCapture(0)  # Camera index 0

# CUDA-based preprocessing functions
def preprocess_frame_cuda(frame):
    # Convert to grayscale (non-CUDA for now, OpenCV doesn't offer cuda.cvtColor)
    gray = cv2.cvtColor(frame, cv2.COLOR_BGR2GRAY)

    # CUDA: Gaussian Blur
    cu_frame = cv2.cuda_GpuMat()
    cu_frame.upload(gray)
    gaussian_filter = createGaussianFilter(cv2.CV_8UC1, cv2.CV_8UC1, (5, 5), 0)
    blurred = gaussian_filter.apply(cu_frame)

    # CUDA: Canny Edge Detection
    canny = createCannyEdgeDetector(50, 150)
    edges = canny.detect(blurred)

    # Download result back to CPU
    return edges.download()

while cap.isOpened():
    ret, frame = cap.read()
    if not ret:
        break

    # Resize frame for faster CUDA processing
    frame = cv2.resize(frame, (640, 480))  # Optional CPU resize for consistent inputs

    # Step 1: Preprocess using CUDA
    edges = preprocess_frame_cuda(frame)

    # Step 2: Resize edges for PyTorch model
    cu_edges = cv2.cuda_GpuMat()
    cu_edges.upload(edges)
    resized_edges = resize(cu_edges, (160, 80), interpolation=cv2.INTER_LINEAR)
    input_frame = resized_edges.download() / 255.0  # Normalize

    # Step 3: Convert to PyTorch Tensor
    input_tensor = torch.from_numpy(input_frame).float().unsqueeze(0).unsqueeze(0).to(device)  # Shape: (1, 1, 80, 160)

    # Step 4: Predict with PyTorch model
    with torch.no_grad():
        predicted_mask = model(input_tensor).cpu().numpy()

    # Step 5: Postprocess output (convert to binary mask)
    output = (predicted_mask[0, 0] > 0.5).astype(np.uint8) * 255
    output_resized = cv2.resize(output, (frame.shape[1], frame.shape[0]))
    overlay = cv2.addWeighted(frame, 0.8, cv2.cvtColor(output_resized, cv2.COLOR_GRAY2BGR), 0.2, 0)

    # Step 6: Display results
    cv2.imshow("Lane Detection (CUDA Optimized)", overlay)

    # Exit when 'q' is pressed
    if cv2.waitKey(1) & 0xFF == ord('q'):
        break

cap.release()
cv2.destroyAllWindows()
```

evertonsmotta, 2025-02-03:
OPENCV insllad for CPP and PYthon.

pkg-config --modversion opencv4
4.1.1

----

pip show opencv-python
Name: opencv-python
Version: 4.11.0.86
Summary: Wrapper package for OpenCV python bindings.
Home-page: https://github.com/opencv/opencv-python
Author: 
Author-email: 
License: Apache 2.0
Location: /home/okdot5/.local/lib/python3.10/site-packages
Requires: numpy
Required-by:

matde-je, 2025-02-03:
```python
import cv2
import numpy as np

def create_binary_mask(image_path, annotation_path):
    """
    Creates a binary mask from CULane annotations.

    Args:
        image_path: Path to the image file.
        annotation_path: Path to the annotation file.

    Returns:
        A NumPy array representing the binary mask.
    """

    img = cv2.imread(image_path)
    height, width = img.shape[:2]
    mask = np.zeros((height, width), dtype=np.uint8) 

    with open(annotation_path, 'r') as f:
        lines = f.readlines()
        for line in lines:
            # Parse the line to extract x-coordinates 
            # (assuming a specific format in the annotation file) 
            x_coords = [float(x) for x in line.strip().split()] 
            y_coords = np.arange(0, height)  # Assuming y-coordinates are evenly spaced

            # Draw the lane line on the mask
            for x, y in zip(x_coords, y_coords):
                if x >= 0:  # Ignore invalid points
                    cv2.circle(mask, (int(x), int(y)), radius=2, color=255, thickness=-1) 

    return mask

```

evertonsmotta, 2025-02-03:
Path for CPP tests on NANO:

~/Documents/e-code/opencv_tests/cpp

matde-je, 2025-02-05:
# Reading data real time through opencv, preprocessing and sending it to the model

```python
import torch
import torch.nn.functional as F
import cv2
import numpy as np
from model import LaneNet  # Your trained model
import torchvision.transforms as transforms

# Load the trained model
device = torch.device("cuda" if torch.cuda.is_available() else "cpu")
model = LaneNet().to(device)
model.load_state_dict(torch.load('lanenet_model.pth', map_location=device))
model.eval()

# Define OpenCV video capture (0 = default webcam, or replace with video file)
cap = cv2.VideoCapture(0)  # Change to "video.mp4" for a video file

# Define preprocessing
transform = transforms.Compose([
    transforms.ToPILImage(),
    transforms.Resize((256, 256)),  # Resize to match the model's training size
    transforms.ToTensor()
])

while cap.isOpened():
    ret, frame = cap.read()
    if not ret:
        break

    # Preprocess the frame
    image = cv2.cvtColor(frame, cv2.COLOR_BGR2RGB)  # Convert to RGB
    image = transform(image).unsqueeze(0).to(device)  # Add batch dimension

    # Forward pass through the model
    with torch.no_grad():
        output = model(image)  # Get model prediction
        pred_mask = torch.sigmoid(output).cpu().squeeze().numpy()  # Convert to numpy

    # Convert to binary mask (Threshold at 0.5)
    pred_mask = (pred_mask > 0.5).astype(np.uint8) * 255  # Convert to binary (0 or 255)

    # Resize the mask to original frame size
    pred_mask = cv2.resize(pred_mask, (frame.shape[1], frame.shape[0]))

    # Overlay mask on the frame
    overlay = frame.copy()
    overlay[pred_mask > 0] = [0, 255, 0]  # Color detected lanes in green

    # Display the frame
    cv2.imshow("Lane Detection", overlay)

    # Press 'q' to exit the real-time display
    if cv2.waitKey(1) & 0xFF == ord('q'):
        break

cap.release()
cv2.destroyAllWindows()
```

JoelenCruz, 2025-02-25:
```python
import cv2
import numpy as np
import torch

def get_lane_center(mask):
    """
    Finds the lane center from the segmentation mask.
    """
    # Convert tensor to numpy array (assuming single-channel mask)
    mask = mask.cpu().numpy().squeeze().astype(np.uint8)  # (H, W)

    # Get bottom half of the mask (focus on road area)
    height, width = mask.shape
    bottom_half = mask[int(height / 2):, :]

    # Find lane pixels
    lane_pixels = np.where(bottom_half > 0)

    if len(lane_pixels[0]) == 0:
        return None  # No lane detected

    # Compute average x-position of lane pixels (center of lane)
    lane_center_x = int(np.mean(lane_pixels[1]))
    image_center_x = width // 2

    return lane_center_x, image_center_x

def compute_steering_angle(lane_center_x, image_center_x, max_angle=30):
    """
    Compute steering angle based on lane center deviation.
    """
    if lane_center_x is None:
        return 0  # No lane detected, keep current direction

    # Compute deviation (negative = left, positive = right)
    error = lane_center_x - image_center_x

    # Normalize error (-1 to 1)
    error_normalized = error / (image_center_x)

    # Convert to steering angle (-max_angle to max_angle)
    steering_angle = max_angle * error_normalized

    return np.clip(steering_angle, -max_angle, max_angle)

from jetracer.nvidia_racecar import NvidiaRacecar

car = NvidiaRacecar()
car.throttle = 0.2  # Set speed (adjustable)

def drive(predicted_mask):
    lane_center_x, image_center_x = get_lane_center(predicted_mask)
    steering_angle = compute_steering_angle(lane_center_x, image_center_x)

    # Normalize steering to range (-1, 1) for JetRacer
    car.steering = steering_angle / 30  # Normalize to [-1,1]

    print(f"Steering: {car.steering}, Lane Center: {lane_center_x}")

```

closed 2025-08-28 by matde-je